JitterCompany/f1-led-circuit #1: Select Microcontroller

CodingInGreen, 2024-04-01:


CodingInGreen, 2024-04-01:
I have experience getting Wifi connectivity working on an ESP32C3, which also supports  2.4 GHz Wi-Fi (802.11 b/g/n) and Bluetooth Low Energy (BLE) 5.0.

CodingInGreen, 2024-04-02:
probe-rs issue resolved, problem was caused by outdated probe-rs version. Board could be a good candidate.

CodingInGreen, 2024-04-03:
The design is open-source and we have the KiCAD files here:
https://github.com/esp-rs/esp-rust-board

ESP32-C3-MINI-1 comes with a PCB Antenna
ESP32-C3-MINI-1U comes with an external Antenna Connector

Power Supply Voltage:
Min: 3V
Typ: 3.3V
Max: 3.6V

ijager, 2024-04-03:
Does this part work well with RTIC and or Embassy?

CodingInGreen, 2024-04-03:
> Does this part work well with RTIC and or Embassy?

It works well with Embassy, I haven't tried RTIC with this specific MCU.

ijager, 2024-04-03:
Seems like the C3 is the only part in the ESP family that currently works with RTIC (due to the mcu architecture). And they also work with Embassy

I think this is good choice.

CodingInGreen, 2024-04-03:
Found a bug -- https://github.com/rtic-rs/rtic/pull/913

Someone in the chat notified me but I didn't see it until after, of a separate PR that covers it already that hasn't yet been merged -- https://github.com/rtic-rs/rtic/pull/906

haha, seems like we're probably set on RTIC as well. There's an updated example in #906 that I'll test out tomorrow.

CodingInGreen, 2024-04-09:
We've decided to choose the esp32c3 for our MCU.

Reasons why:

- 2.4 GHz Wi-Fi (802.11b/g/n) and Bluetooth® 5 (LE)
- Plays nicely with Embassy
- Configurable clock frequency with a maximum of 120 MHz in Single Transfer Rate (STR) mode (speedy enough for our LEDs)
- Adam has experience with esp32c3 through the dev kit: ESP32-C3-DevKit-RUST-1

CodingInGreen, 2024-04-14:
ESP32-C3-MINI-1 Selected

closed 2024-04-14 by CodingInGreen