JitterCompany/f1-led-circuit #2: Select LED

CodingInGreen, 2024-04-01:


CodingInGreen, 2024-04-01:
https://www.iodparts.eu/part/ws2812b-2020

WS2812B-2020
Worldsemi
SMD 2mm x 2mm
Protocol: NZR communication mode
The control integrated circuit and the LED share the only power source.
12mA operating current per channel.
Stock: 102,785
Days to Ship: 20 Days
Minimum Order Quantity: 4,500 ??

https://www.digikey.pt/pt/products/detail/sunled/XZDGCBDMERK150W-1/15288282?utm_adgroup=&utm_source=google&utm_medium=cpc&utm_campaign=PMax_Product_All%20Products&utm_term=&productid=15288282&utm_content=&utm_id=go_cmp-20187105315_adg-_ad-__dev-c_ext-_prd-15288282_sig-CjwKCAjwtqmwBhBVEiwAL-WAYZHh0PTVpt2g4rryGw_7VLPSQP2KY9bIxFoJDPiLm1TOGtVY-DyeSBoCwzsQAvD_BwE&gad_source=1&gclid=CjwKCAjwtqmwBhBVEiwAL-WAYZHh0PTVpt2g4rryGw_7VLPSQP2KY9bIxFoJDPiLm1TOGtVY-DyeSBoCwzsQAvD_BwE

XZDGCBDMERK150W-1
SunLED
SMD 1 mm x 1 mm
Forward Current: 10mA
Protocol: Not sure if these can even be daisy chained.
Stock: 7,754
Ships Immediately

CodingInGreen, 2024-04-01:
After reviewing: https://www.superlightingled.com/blog/category/addressable-led-strips/addressable-led-ic-comparison/

I'm not 100% if WS2812B-2020 is the best. The 2x2mm form factor is super nice. Each LED draws up to 12mA per channel, so with a 5V/10A power supply that's 10A/36mA = 277.7 LEDs. Taking 80% of that to be conservative, that's 222 LEDs.

ijager, 2024-04-01:
Why is it "the best"? 

Some things to consider: 

* supply voltage
* Is there a supported maximum chained length? 
* What is the maximum refresh rate (fps) for certain lengths of chains? Or maybe in reverse, say we want to achieve 30Hz, how long can we make the chain. (Here you can see why requirements are important to have).

It would be nice to make a nice markdown table with a row for each candidate. So we can compare and make a data driven decision.

CodingInGreen, 2024-04-02:
[led_comparison.xlsx](https://github.com/JitterCompany/f1-led-circuit/files/14832341/led_comparison.xlsx)

Added comparison chart for review.

CodingInGreen, 2024-04-02:
[image]

552 LEDs in total

CodingInGreen, 2024-04-03:
If there are 20 cars on the track at a time, with a maximum current draw of 20 mA per channel with 3 channels. Max draw per RGB LED is 60 mA x 20 cars = Max draw from LEDs for all cars = 1,200 mA or 1.2 A

CodingInGreen, 2024-04-03:
**WS2812B-2020**

The WS2812B-2020 LED utilizes a single-wire non-return-to-zero (NZR) communication mode, making its protocol highly sensitive to timing. This design means that the difference between transmitting a logic '0' or a '1' depends on precise timing intervals, which can introduce complexity in programming, especially on platforms with non-deterministic timing behavior. The protocol's timing sensitivity requires careful management of interrupts and timing functions to achieve stable color patterns and refresh rates, potentially complicating development on multitasking operating systems or environments with variable execution times.

**APA102C**

Conversely, the APA102C employs a two-wire protocol, similar to SPI, using separate data and clock lines, which significantly reduces timing sensitivity and programming complexity. This approach allows for more straightforward and reliable data transmission, making it less prone to errors from timing discrepancies. The inclusion of a clock line not only facilitates higher refresh rates by supporting faster data transmission but also enables the APA102C to operate with a broader range of microcontrollers and software environments, making it a more versatile choice for projects requiring robust and dynamic LED displays.

ijager, 2024-04-03:
This is great. I will inquire about availability and pricing for these parts so we weigh that into the selection.

ijager, 2024-04-03:
For power, assuming only 20 leds are on is too optimistic. We have to be able to handle turning on all LEDs, which can happen because of bugs or when people are going to run their own firmware on the board.

We can add a temperature sensor to monitor the board temperature and shut it down when it gets too hot.

CodingInGreen, 2024-04-04:
I recounted the LEDs for the track on the 300 x 300 pixel grid, and came up with 550 LEDs instead of 552.

550 LEDs x 60mA = 33,000mA or 33 Amps

CodingInGreen, 2024-04-08:
[led_comparison_updated.xlsx](https://github.com/JitterCompany/f1-led-circuit/files/14903831/led_comparison_updated.xlsx)

New Candidates:

SK9822-EC20(SK9822-2020)
APA107-2020
NS107S-2020
HD108-2020
EC20-6812 (SK6812-2020)

CodingInGreen, 2024-04-14:
HD108-2020 selected.

closed 2024-04-14 by CodingInGreen