JitterCompany/f1-led-circuit #5: Select Temperature Sensor

CodingInGreen, 2024-04-11:
I have had a successful experience using an SHT31 temperature and humidity sensor using I2C.

SHT31:

[https://www.digikey.pt/pt/products/detail/sensirion-ag/SHT31-DIS-F2.5KS/6212137?utm_adgroup=&utm_source=google&utm_medium=cpc&utm_campaign=PMax_Product_Zombie%20SKUs&utm_term=&productid=6212137&utm_content=&utm_id=go_cmp-18734772434_adg-_ad-__dev-c_ext-_prd-6212137_sig-Cj0KCQjwlN6wBhCcARIsAKZvD5gSOAvqh30QL-ChqUxGYAhbBT3Ox3EsK55l6X6DxkMbmEh59wAJCwUaAnPtEALw_wcB&gad_source=1&gclid=Cj0KCQjwlN6wBhCcARIsAKZvD5gSOAvqh30QL-ChqUxGYAhbBT3Ox3EsK55l6X6DxkMbmEh59wAJCwUaAnPtEALw_wcB](url)

This is the crate I used:

https://crates.io/crates/sht31

ijager, 2024-04-11:
Is this crate usable in an async project?

https://www.digikey.pt/pt/products/detail/sensirion-ag/SHT31-DIS-F2.5KS/6212137?utm_adgroup=&utm_source=google&utm_medium=cpc&utm_campaign=PMax_Product_Zombie%20SKUs&utm_term=&productid=6212137&utm_content=&utm_id=go_cmp-18734772434_adg-_ad-__dev-c_ext-_prd-6212137_sig-Cj0KCQjwlN6wBhCcARIsAKZvD5gSOAvqh30QL-ChqUxGYAhbBT3Ox3EsK55l6X6DxkMbmEh59wAJCwUaAnPtEALw_wcB&gad_source=1&gclid=Cj0KCQjwlN6wBhCcARIsAKZvD5gSOAvqh30QL-ChqUxGYAhbBT3Ox3EsK55l6X6DxkMbmEh59wAJCwUaAnPtEALw_wcB
This is the crate I used:
https://crates.io/crates/sht31

CodingInGreen, 2024-04-14:
This builds successfully, I didn't connect my SHT31 sensor to the board to test though.

Would you like me to do that to see if I can get a reading?

https://github.com/CodingInGreen/esp32c3-embassy-sht31

CodingInGreen, 2024-04-15:
I will test the SHT31. Maybe the BME280 sensor would be a better choice. 

This crate supports embedded-hal-async as well:
https://crates.io/crates/bme280-rs

I don't have a BME280 on hand, here is it available in Portugal (18,10 euros + shipping is about 5 euros = 23.10 euros). 
https://mauser.pt/catalog/product_info.php?products_id=096-6714

It could arrive in 2-3 days.

CodingInGreen, 2024-04-15:
I had some trouble getting the readings on esp32c3, so I tested on the micro:bit V2. 

I got an average time measurement: timestamp of data collection cycle - timestamp of new reading = 98 milliseconds.

The micro:bit V2 has a clock speed of 64 Mhz and the measurement frequency can be set up to 400 Mhz (limited by the IC2 setup on the micro:bit V2 crate).

The clock speed of esp32-c3-mini-1 is 160 Mhz, and the maximum measurement frequency of the SHT31 is 1 Mhz.

Calculate for estimate for the blocking time on the esp32-c3-mini-1:

CPU speed increase from 64 MHz to 160 MHz (a factor of 2.5)
I2C speed increase from 400 kHz to 1 MHz (a factor of 2.5)

Time reduction = 2.5 x 2.5 = 6.25

98 milliseconds (avg. time from micro:bit V2) / 6.25 = 15.68 milliseconds

Estimated average blocking time - around 15 milliseconds

CodingInGreen, 2024-04-18:
We went with MCP9701A as it's been successfully used in other projects, and has a margin of error that is acceptable.

closed 2024-04-18 by CodingInGreen