JitterCompany/f1-led-circuit #21: Create basic HD108 example

CodingInGreen, 2024-06-14:
Requirement - Turn one HD108-2020 LED of the f1-led-circuit board a specific RGB color.

CodingInGreen, 2024-06-15:
I found this documentation regarding migration from embedded-hal 0.2 to 1.0 regarding SpiDevice and SpiBus.

https://github.com/rust-embedded/embedded-hal/blob/master/docs/migrating-from-0.2-to-1.0.md#spi-busdevice-separation

CodingInGreen, 2024-06-15:
I worked on ensuring the correct traits from embedded-hal and embedded-hal-async were implemented. This included creating a wrapper around the spi instance to adapt it to the SpiBus trait required by the HD108 driver. It seems there is a mismatch between the expected traits between embedded-hal-async and the esp-hal library. The HD108 driver required an implementation of the SpiBus trait from embedded-hal-async, but the SPI implementation in esp-hal did not directly conform to this trait. The SpiBus trait has methods like read, write, transfer, transfer_in_place, and flush, which are asynchronous. The esp-hal library provides synchronous methods that needed to be wrapped or adapted. I had many attempts but was unable to successfully create a working wrapper.

ijager, 2024-06-16:
You cannot really wrap a sync API in order to make it _async_. Because for async you need a WAKER that can notify the async runtime that it is done with the `await` part.

CodingInGreen, 2024-06-16:
Can we use this? It has a SpiBus trait. 

https://docs.embassy.dev/embassy-embedded-hal/git/default/adapter/index.html

ijager, 2024-06-17:
Sure for a first demo, definitely

closed 2024-06-18 by CodingInGreen