JitterCompany/f1-led-circuit #42: Remove unnecessary components

ijager, 2024-08-03:
- JTAG connector (But perhaps we can expose these pins with some test points on the backsize
- JP1 jumper: same, expose with testpoints
- Boot button: expose pin via test-point

CodingInGreen, 2024-08-08:
I added changed those items to test points in the schematic, can you check them and let me know if they are correct?

CodingInGreen, 2024-08-08:
I've updated the PCB layout, just pushed the changes. 

I'm not sure if there is a standard order for the test points to be in, or if the layout order for the test points matters?

Can you take a quick look when you have a chance?

ijager, 2024-08-08:
This looks an extra GND and 3V3 testpoint to the JTAG series

CodingInGreen, 2024-08-08:
Ok added! Let me know if we can close this issue please. Thanks!

closed 2024-08-08 by ijager