JitterCompany/f1-led-circuit #48: Improve component placement for improved aesthetics

ijager, 2024-08-03:
Ideally the non-LED components could represent some buildings around the track.

- USB C connector should probably still face downwards

CodingInGreen, 2024-08-08:
How about the power components, do they need to be close together near the USB-C Connector?

ijager, 2024-08-08:
Since we decided to put those on the back I suggest to keep them close together so that a cutout in the backplane can remain small

CodingInGreen, 2024-08-08:
There are quite a few capacitors that will live inside the LED race track, if we choose to utilize them. 

What are your thoughts on whether we will have the boards built with capacitors already on the boards, or just the pads?

ijager, 2024-08-08:
Maybe we can remove them or place them on the bottom

CodingInGreen, 2024-08-08:
With a 4 layer PCB I'd like to place them on the bottom and try it, sound okay?

CodingInGreen, 2024-08-09:
Actually if we can remove them we can save on components cost.